jpconstantineau/Community_nRF52_Arduino #19: Readme update

ddB0515, 2021-06-09:
Hi

I'm planing to prepare board based on nRF52840 and make PR to be available with  Community_nRF52_Arduino but Readme is lacking information in general but most how to add new boards, suggestions, etc... 

Recently I have requested PID for firmware/bootloader so can be used with Adafruit Bootloader and submitted as you are for openmoko https://github.com/openmoko/openmoko-usb-oui/pull/29

as you have done same for https://github.com/openmoko/openmoko-usb-oui/pull/20
and as you can check there are some missing information for Licensing also

So if you can check and add description that 3rd party libraries using different licenses 

Here is list that I have managed to get (if I missed some add it to the list please)

# Licenses
- Communitiy nRF52 - Lesser General Public License v2.1
- Adafruit_LittleFS - BSD 3-Clause
- BlueFruit - MIT License
- Adafruit_TinyUSB_Arduino - MIT License
- BLEAdafruitService - MIT License
- BLEHomeKit - BSD License
- BlueFruit52 - MIT License
- InternalFileSytem - MIT License
- PDM - Lesser General Public License v2.1
- RotaryEncoder - MIT License
- SPI - Lesser General Public License v2.1
- Servo - Lesser General Public License v2.1
- SoftwareSerial - Lesser General Public License v2.1
- Wire - Lesser General Public License v2.1

jpconstantineau, 2021-06-10:
If you do a PR, can you do it on the develop branch?  there has been quite a few changes since the last push to main.  I suspect that I will need to do a large update before I start at looking at merging in the changes the 0.22.x adafruit nrf52 releases have brought in.

ddB0515, 2021-12-29:
Hello @jpconstantineau 

Sorry for long time not responding as life happening  and too much distraction
I'll create PR soon as I just submitted PR for Adafruit Bootloader
https://github.com/adafruit/Adafruit_nRF52_Bootloader/pull/258

Also have stashed changes for board locally so need to make checks before I do it but expect soon
Maybe if I manage this year if not next one ;)

jpconstantineau, 2021-12-29:
> adafruit/Adafruit_nRF52_Bootloader#258

Wow.  It's been a while indeed.

You can do a PR for your changes and I'll bring them in.
You might do a PR in 2 places as I am thinking of having a proper fork of
the Adafruit nrf52 BSP
you can find it here:

jpconstantineau, 2021-12-29:
Wow.  It's been a while indeed.

You can do a PR for your changes and I'll bring them in.
You might do a PR in 2 places as I am thinking of having a proper fork of
the Adafruit nrf52 BSP
you can find it here:

https://github.com/nRF52-Arduino/nRF52_Arduino/tree/main

This is in addition to the outdated community nrf52 bsp.
It's outdated as it's from prior 0.20 I believe - where they had issues
with high power consumption.

I am slowly working on my BlueMicro firmware to support the latest BSP from
adafruit, once that is done, I'll upgrade the community BSP.

Pierre