jpconstantineau/Community_nRF52_Arduino #20: PlatformIO IDE support

rickygai, 2021-11-26:
I cannot find any board based on the Nordic nRF52840 Dongle under PlatformIO IDE.

How to put in the link "https://github.com/jpconstantineau/Community_nRF52_Arduino/releases/latest/download/package_jpconstantineau_boards_index.json" under PlatformIO IDE ?

Please advise.

jpconstantineau, 2021-11-26:
I have never used PlatformIO.  Unfortunately, for this Google will be the guide... Even myself would have to google it...

rickygai, 2021-11-27:
ic, you may try PlatformIO IDE is far more better than Arduino IDE either 2.0 beta or 1.8.16

Thanks for the feedback.

maxgerhardt, 2021-11-27:
I've created a PlatformIO project that uses the 0.21.0 version of this core (since I could *not* get the master branch to compile due to not knowing which Adafruit core it was based on) at https://github.com/maxgerhardt/pio-community-nrf52-example. 

It compiles but I'm unsure about the upload method. What upload method do you use in the Arduino IDE for this board? The serial port (and thus nrfutil) or JLink?

jpconstantineau, 2021-11-27:
I use bootloader DFU for all the boards (which is serial).  I only use the jlink when I need to flash the bootloader.

maxgerhardt, 2021-11-27:
I see. I've changed the default upload method to `nrfutil` then which attempts uploading via 

```
Auto-detected: COM11
Forcing reset using 1200bps open/close on port COM11
Waiting for the new upload port...
"C:\Users\Max\.platformio\packages\framework-arduinoadafruitnrf52\tools\adafruit-nrfutil\win32\adafruit-nrfutil.exe" dfu serial -p COM11 -b 115200 --singlebank -pkg .pio\build\pca10059\firmware.zip
Upgrading target on COM11 with DFU package C:\Users\Max\temp\jpnrf\.pio\build\pca10059\firmware.zip. Flow control is disabled, Single bank, Touch disabled
```

which I hope is correct in principle. 

@rickygai Please test the above project regarding compilation & uploading.

rickygai, 2021-12-01:
Hi maxgerhardt, 

_"I've created a PlatformIO project that uses the 0.21.0 version of this core"_
- may i know how to do this, where to edit configuration ?

_"Please test the above project regarding compilation & uploading."_
- currently under Arduino IDE 1.8.16, I am using serial port to upload after new bootloader is written, so far is working fine.
- but under platformio ide the "variant.h" is not based on:
           
           C:\Users\<name>\AppData\Local\Arduino15\packages\community_nrf52\variant.h
           
  The variant.h platformio ide is based on:
  
          C:\.platformio\packages\framework-arduinoadafruitnrf52\variants\feather_nrf52840_express\variant.h
  
I can compile and upload under platformio, but the pins assigned are wrong because of wrong variant.h used.

I am new to platformio, I wondered how to add the Community add-on board of Nordic nRF52840 Dongle (PCA10059) to the board listing of platformio. 

Thank you jpconstantineau for creating this add-on board, at least I still can work under Arduino IDE for the moment.

Please advise.

maxgerhardt, 2021-12-01:
> may i know how to do this, where to edit configuration ?

Configuration meaning what exactly? The [pca10059.json](https://github.com/maxgerhardt/pio-community-nrf52-example/blob/main/boards/pca10059.json) is where the main configuration happens regarding the chosen variant and build flags, the other one being [the platformio.ini](https://github.com/maxgerhardt/pio-community-nrf52-example/blob/main/platformio.ini#L14-L20) regarding the core version.

>but under platformio ide the "variant.h" is not based on: `C:\Users\<name>\AppData\Local\Arduino15\packages\community_nrf52\variant.h`

1. PlatformIO **never** uses any global Arduino IDE installation, 2. no the "variant.h" ArduinoIDe uses is located in `C:\Users\<user>\AppData\Local\Arduino15\packages\community_nrf52\hardware\nrf52\0.1.21\variants\pca10059` in when you chose the ["Nordic nRF52840 Dongle (PCA10059)"](https://github.com/jpconstantineau/Community_nRF52_Arduino/blob/main/boards.txt#L318-L319) board.

>The variant.h of platformio ide is based on: `C:\.platformio\packages\framework-arduinoadafruitnrf52\variants\feather_nrf52840_express\variant.h`

No, not with the project I've created. When I look into the `.vscode/c_cpp_properties` and the compilation log, it's using the correct `C:/Users/<user>/.platformio/packages/framework-arduinoadafruitnrf52/variants/pca10059` folder which is exactly the folder specified in the board's JSON manifest. Are you sure you've used the [project environment switcher](https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks) to change your active project to the correct one, did at least a build **once** and used Ctrl+Shift+P -> Rebuild IntelliSense to update the C/C++ autocompletion index?

rickygai, 2021-12-01:
_"No, not with the project I've created. W..."_
- actually i created the folder "c:\.platformio" manually before installing the platformio.

Is ok, will google and see if there are some info out there.

rickygai, 2021-12-01:
Hi maxgerhardt, 

I manage to compile and upload to nRF52840 Dongle via serial port, and it is functioning properly now.

One thing I did not change is, the [env:pca10059] where remained as [env:adafruit_feather_nrf52840] not sure any issues with this but so far I tested the pins are correctly mapped. 

The entire process is below:

[image]

Thank you.

maxgerhardt, 2021-12-01:
>One thing I did not change is, the [env:pca10059] where remained as [env:adafruit_feather_nrf52840] not sure any issues with this but so far I tested the pins are correctly mapped.

The `[env:x]` name is not relevant, this just affects how the folder in which the compiled files are stored is named; the build settings inside the environment is what matters.

One thing I would not is that there needs to be proper indentation for the `platform_packages` instruction
```ini
; bad
platform_packages = 
framework-arduinoadafruitnrf52@...
; good
platform_packages = 
   framework-arduinoadafruitnrf52@...
```
And that copying the board definition in the platform folder is only needed if you need global access to that board definition. A newly created project will still need that `platform` and `platform_packages` modification. The way I set-up the project was that it's completely self-contained through the project having a `boads/` folder, no need to manually copy files. But as said, your method is valid if you don't want to copy that per-project.

maxgerhardt, 2021-12-01:
Support for this modified core on the PlatformIO side is now tracked in https://github.com/platformio/platform-nordicnrf52/issues/134.