jpconstantineau/Community_nRF52_Arduino #22: Add support for nrf52833

carrino, 2022-06-04:
Add support for debug board PCA as well as the bluemicro board.  Ideally add support for softdevice 7.2 as well as 6.1.1

jpconstantineau, 2022-07-25:
Good news, Adafruit has starting working on getting support for the nrf52833.  See this PR: https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/739

Once that is merged, I will be able to get support for the nrf52833 included in the BSP without much issue. (It's just a matter of time)

h2zero, 2022-08-07:
Please forgive the promo, I just stumbled across this and thought I'd share an alternative that does not rely on a softdevice or adafruit and supports generic (any nrfx, including 52833) boards using NimBLE.  You can check it out here: https://github.com/h2zero/n-able-Arduino