jpconstantineau/Community_nRF52_Arduino #24: Design of nRF52840 missleadings

ITstreet1, 2022-12-02:
Hi,
I am trying to design a board based on BlueMicro using this schematic:
[http://nrf52.jpconstantineau.com/assets/images/schematic_bluemicro840_1.0-a8bab4054d9ddeeb92fb4297b6b22eda.png](url)

While checking variants and pins for this board here:
[https://github.com/jpconstantineau/Community_nRF52_Arduino/blob/main/variants/bluemicro_nrf52840/variant.h](url)

I noticed some wrong pin mappings. 

In Variants you say RXD is:
#define PIN_SERIAL1_RX      _PINNUM(1, 13)
which should be P1.13, but on the schematic, it is P0.08. Which one is correct?

Besides, If someone has time to give a hand with this design, it would be appreciated. Just to check if all is ok when I am done.

Regards

jpconstantineau, 2022-12-03:
That's a good point.  Since the nrf52 can use any GPIO for serial, this can be remediated pretty easily in firmware.

ITstreet1, 2022-12-03:
Ok.
So, correct me where I am wrong.

If I use your lib, the default pins for interfaces should be:

SPI:
MISO - P0.09
MOSI - P1.11
SCK    - P0.10
SS      - P0.03

UART:
RX - P1.13
TX - P0.28

I2C:
SDA - P0.30
SCL - P0.02

LED1 - P1.04
LED2 - P1.10

BTN1 - P0.18
BTN2 - P1.01

NFC1 - P0.09
NFC2 - P0.10

How to address analog and digital pins?
From what I found here:
https://bluemicro.jpconstantineau.com/docs/understand_gpios
digital shouldn't be a problem. Port 0 should be the number of bits, and for port 1 just add 32 to the bit number. So P0.26 should be D26 and P1.02 should be D34.
What about analog? Say A2? If I understood you correctly, analog Arduino mappings do not follow AI0-7.

jpconstantineau, 2022-12-04:
> [image: Schematic_GREENboard nRF52840_2022-12-04]
> <https://user-images.githubusercontent.com/30090189/205490522-eeb88ec9-a01a-48f8-9b9e-142ab51215fc.png>
>
> How this looks up? Did I miss something?
> Voltage regulators and pins will come later.

From a brief glance, looks ok. I didn't see the USB lines to a connector

I don't believe DFU is setup in my bootloader. If you want to use it, you
will have to make sure it's coded properly (pins set up) in the bootloader
code. This wouldn't be in the board support package as DFU is only used by
the bootloader, which runs before the "Arduino" code runs...

As for pin functions to gpios, I didn't check that as only the crystal and
the analog pins cannot be moved. Everything else can move around.

I believe 06 and 08 for serial might be legacy pins from the 832
bootloader. They can be remapped in firmware.

ITstreet1, 2022-12-05:
USB I didn't put here to make the schematic clear. There are D- and D+.

As for DFU, I might need it in the future. 

So, using this project, your Bootloader, it should work?

ITstreet1, 2022-12-06:
Thank you.

PCBs are on their way. I will let you know how it turned up. :)